MAPIRlab/gaden #2: Folder missing "logs_gas_dipersion_simulator"

jbourne15, 2017-07-13:
Hey BTW the repository is missing the folder "logs_gas_dipersion_simulator".

Also, how do you get wind data "wind_at_cell_center_points_0.csv_U" from the CFD results U.gz? I am new to CFD so sorry in advance if this is straight forward. 

thanks for the help,
Joseph

jbourne15, 2017-07-13:
Also, is the OccupancyGrid3D.csv file generated from the mesh used in CFD?

JGMonroy, 2017-07-18:
Hi jbourne15,
I answer your questions in order:

1. The "logs_gas_dipersion_simulator" is a folder not included in the repo since it is a configurable folder where to store the simulations. It can be set from the launch file using the parameter "results_location", see the [gaden_simulator_demo.launch](https://github.com/MAPIRlab/gaden/blob/master/gaden_demo/launch/gaden_simulator_demo.launch).

2. Regarding how to process the wind simulations from CFD in order to be used in GADEN, I'm aware this is not a trivial step. I'm preparing a "tutorial" with some instructions to ease the process of generating new environments. I hope to release it before August. Sorry for the delay.

jbourne15, 2017-07-18:
Awesome! I am so glad I found your paper on this work; I was considering implementing something similar.

I am assuming you will be doing a "tutorial" for the OccupancyGrid3D.csv as well?

JGMonroy, 2017-07-18:
Yes, everything will be documented.

JGMonroy, 2017-07-19:
Hi again,
I've uploaded to the "[gaden_demo](https://github.com/MAPIRlab/gaden/tree/master/gaden_demo/demo)" pkg a tutorial explaining how I generate the CAD models, run the CFD simulations and then post-process the data to generate the occupancy3D.csv and the wind flow vectors U,V,W for the environment (inputs for GADEN).

Please, tell me if there is something missing.
Best,
Javier Monroy

jbourne15, 2017-07-19:
hello, great I will let you know if I find anything.

jbourne15, 2017-07-31:
Hello Javier Monroy,

I am about finished with your tutorial and I was wondering about whether or not time varying wind flow data is functional in your code yet? It seems to be, but I haven't yet tried it yet or tried post processing the CFD data.  

Thanks, 
Joseph Bourne

JGMonroy, 2017-07-31:
Hello,

There should be no problem. The varying wind flow data is programmed in a
similar way as the gas dispersal (which is also varying). The only thing is
to post-process not only one CFD time-step, but multiple.

Javier G. Monroy
PostDoc Researcher at MAPIR <http://mapir.isa.uma.es/> lab
University of Malaga
http://mapir.isa.uma.es/jgmonroy

JGMonroy, 2017-08-02:
Since the question on this issue seems to be solved, I'm closing it.

closed 2017-08-02 by JGMonroy