MAPIRlab/gaden #3: GasSimulator.h missing?

Huitzilo, 2017-08-02:
Hi,

when trying to compile GADEN I get stuck with the following error:

`In file included from /<...>/gaden/gaden_player/src/simulation_player.cpp:8:0:gaden/gaden_player/src/simulation_player.h:9:38: fatal error: gaden_player/GasPosition.h: No such file or directory`

There doesn't seem to be a GasPosition.h in the repository, only GasPosition.srv at `gaden/gaden_player/srv/GasPosition.srv`.

Am I missing a step in the preparation, or another repo to include? 

Thanks in advance,

Michael

JGMonroy, 2017-08-02:
Hi Huitzilo,
The GasPosition.h is a file that should be auto-generated when compiling the "gaden_player" pkg (see [here](http://wiki.ros.org/ROS/Tutorials/WritingServiceClient%28c%2B%2B%29) for example). 

As you mention, the GasPosition is a service (you can see that the CMakeLists.txt includes the "message_generation" and the .srv file), and therefore the *.h files should be generated by catkin when compiling it.

The only thing I've detected so far, is that the "package.xml" is missing the necessary headers (I've committed the changes):
`<build_depend>message_generation</build_depend>
  <run_depend>message_runtime</run_depend>`

Huitzilo, 2017-08-02:
Hi Javier,

OK, that solved it. Previously I was calling `catkin_make` on the whole workspace when it failed. I had to run `catkin_make --pkg gaden_player` first to build that package. After that calling catkin_make on the whole workspace worked.

Thanks for your help!

Cheers,

Michael

closed 2017-08-03 by Huitzilo