MAPIRlab/gaden #4: MOX sensor readings

jbourne15, 2017-08-17:
I was wondering if you could explain the response of the simulated MOX sensors specifically, why does the concentration (in ppm) go down when the sensors are within the emitting filaments?

[image]

JGMonroy, 2017-08-22:
Hi jbourne15,
I think the misunderstanding is that the sensors are providing readings in [Ohm] not in [ppm], therefore when the gas concentration increases, the sensor resistance decreases, as seen in your plot. That is, GADEN provides measures in [ppm], but when you simulate a MOX sensor you obtain the sensor readings based on the concentration they are exposed to. For the case of MOX sensors (you can see the manufacturer's datasheets) the readings are based on the sensor resistance changes, therefore the value you obtain by simulating a MOX sensor is the sensor resistance (Ohms).

If you want to see what is the concentration in (ppm), you can simulate a PID sensor, which provides measurements in (ppm) rather than resistances.

jbourne15, 2017-08-22:
Thank you for this clarification, I was under the impression that the simulated MOX sensors reported ppm.

closed 2017-08-22 by jbourne15