MAPIRlab/gaden #7: invalid quaternions

r-miko, 2018-01-17:
Hello,

When I run the launch file gaden_simulator_demo.launch I get this error:

[image]

Have I missed something?

Thanks in advance,
Rebecca

JGMonroy, 2018-02-01:
Hi r-miko,
Is this problem related to the solved [issue 8](https://github.com/MAPIRlab/gaden/issues/8)??.
If not, the problem is related to the visualization of some markers on rviz (the gas sources) which are defined in the node "environment". They do not affect the simulation, but I'll take a look after the latest commits if the problem persists.

r-miko, 2018-02-06:
Hello Javier,

It must have been related to #8 as the problem is now gone. 
Thanks again!

closed 2018-02-06 by JGMonroy