MAPIRlab/gaden #8: gaden_demo error

r-miko, 2018-01-17:
Hello,

When I run the launch file gaden_simulator_demo.launch I get the error: 
_Error in opening the final output files._
When I exit, it tries to write to the log file "filament_simulator1-3*.log", but I this file is not in the log directory. 
After running the debugger gdb on the filament simulator, I get:

[image]

Also, this error may or may not be related to https://github.com/MAPIRlab/gaden/issues/7 

Thank you in advance for your help,
Rebecca

Huitzilo, 2018-02-01:
Hi, I get this error, too... That log file doesn't exist, and the other log files in the log directory don't give away any clue about what's going wrong (except that warning about unnormalised quaternions from #7 , which doesn't really help).

We're really super-keen to replicate the results from the paper. Any ideas how to start debug this?Thanks in advance!

Huitzilo, 2018-02-01:
Solved it: 
`mkdir /<path_to_catkin_workspace>/src/gaden/gaden_demo/demo/logs_gas_dipersion_simulator/`

@r-miko

JGMonroy, 2018-02-01:
Hi, 
This problem seems to be related to being unable to save the output log files with the simulation. I'll take a look at [pull request #9 ](https://github.com/MAPIRlab/gaden/pull/9) as soon as possible to create the folder if not found, but also take into account that you can change the path where to save the logs with the "/results_location" param set in the launch file:
`<param name="/results_location" value="$(find gaden_demo)/demo/logs_gas_dipersion_simulator" />`

Huitzilo, 2018-02-01:
Hi Javier,

the pull request simply creates the results directory in the github tree, so a pristine checkout should work just fine since the results directory is already in place. Thanks for the hint towards the /results_location parameter, might come in handy, too!

JGMonroy, 2018-02-01:
Hi Huitzilo,
Your pull request has been merged (thanks). I did not create the folder at first hand because I thought it would be better to separate the code from the results, but maybe for a first contact its better not to have errors with the demo XD. The correct way should be to set the `/results_location` parameter as described in the launch file.

r-miko, 2018-02-02:
Hi,
This works for me too. Thank you!