MAPIRlab/gaden #11: map error

r-miko, 2018-05-09:
Hello,

When I run the launch file gaden_player_demo.launch I get the error:

[image]

Thank you in advance for your help,
Rebecca

hdino, 2018-09-06:
@r-miko I encounter the same problem. Did you find a solution?

GUYING7, 2018-12-02:
@hdino
Hi,hdino! I encounter the same problem. Did you find a solution?

JGMonroy, 2019-07-03:
GADEN_demo has not been replaced by /test_env folder with different test environments. All the new scenarios have been tested and launch files provided.

closed 2019-07-03 by JGMonroy

velonSigs, 2023-10-31:
> Hello,
> 
> When I run the launch file gaden_player_demo.launch I get the error:
> 
> [image]
> 
> Thank you in advance for your help, Rebecca

hi,how do you solve this problem? I also met this issue