MAPIRlab/gaden #14: Launch GADEN_preprocessing.launch

rafluqber95, 2019-07-07:
Hi! I would like to run any of the examples of the test_env folder, but I don't know how to do it very well because I get this error when I launch the GADEN_preprocessing.launch file:
("roslaunch /home/usuario/catkin_ws/src/gaden-master/test_env/10x6_central_obstacle/launch/GADEN_preprocessing.launch") 

ERROR: cannot launch node of type [gaden_preprocessing/preprocessing]: can't locate node [preprocessing] in package [gaden_preprocessing]

(Or when I do: "roslaunch test_env GADEN_preprocessing.launch" 
I get this error: 

multiple files named [GADEN_preprocessing.launch] in package [test_env]:
- /home/usuario/catkin_ws/src/gaden-master/test_env/10x6_snake/launch/GADEN_preprocessing.launch
- /home/usuario/catkin_ws/src/gaden-master/test_env/10x6_central_obstacle/launch/GADEN_preprocessing.launch
- /home/usuario/catkin_ws/src/gaden-master/test_env/MAPIRlab/launch/GADEN_preprocessing.launch
- /home/usuario/catkin_ws/src/gaden-master/test_env/10x6_empty_room/launch/GADEN_preprocessing.launch
- /home/usuario/catkin_ws/src/gaden-master/test_env/10x6_maze/launch/GADEN_preprocessing.launch
Please specify full path instead
The traceback for the exception was written to the log file
)

I only installed the complete project into my catkin_ws/src folder and compile it.

Anybody could tell me any advice? Thank you very much, and sorry for the inconveniences.

JGMonroy, 2019-07-08:
Hi @rafluqber95,

I answer you in reverse order:

-  `roslaunch test_env GADEN_preprocessing.launch` --> As you point out, this method cannot be used as I replicated the name of the launch files for every test scenario in the same **test_env** pkg. Therefore, rising this issue of multiple files with the same name (I thought it was a good idea to have all launch files with the same name as to easily introduce new scenarios, but maybe I've to think it again.)

- The correct way is to use the full path to the launch file so that ROS does not have to mess with multiple files named the same. For example: `roslaunch /home/<username>/catkin_ws/src/olfaction/gaden/test_env/10x6_central_obstacle/launch/GADEN_preprocessing.launch`. The latter works for me with no problems. 

To check that GADEN is properly installed, confirm that:
- You have sourced the correct `catkin_ws` in your `.bashrc` file.
- rospack can find GADEN and test_env pkgs: `rospack find gaden_preprocessing`, `'rospack find test_env`.
- That `catkin_make` does run without errors.

Let me know your outputs.
Best,
Javier

JGMonroy, 2019-07-08:
Also, remember that for some GADEN components, you need to download some *msg* definitions from the repo [olfaction_msgs](https://github.com/MAPIRlab/olfaction_msgs).

rafluqber95, 2019-07-08:
> Also, remember that for some GADEN components, you need to download some _msg_ definitions from the repo [olfaction_msgs](https://github.com/MAPIRlab/olfaction_msgs).

Now yes, thank you very much!

closed 2019-07-08 by rafluqber95

forgetto, 2023-03-16:
Hello, this is the first time for me to use GADEN. I am not very clear about its installation and use, so I would like to ask if there are any documents or video tutorials, thank you very much