MAPIRlab/gaden #15: "map" passed to lookupTransform argument target_frame does not exist.

kirk0306, 2019-08-01:
Dear developers,
Hi!

When I run the launch file `roslaunch GADEN_player.launch` I get the error and the rpt_plot is blank . 

I think GADEN is properly installed,  because `catkin_make` was run without errors. There is only one scenario named _10x6_snake_ in the file _/test_env_, I had tried to run the below commands and got outputs:

`roslaunch GADEN_preprocessing.launch`
..
[ INFO] [1564687470.057891299]: Preprocessing done

`roslaunch GADEN.launch`
..
[ WARN] [1564687479.027664857]: [filament] File /home/karuka/catkin_ws/src/olfaction/gaden-master/test_env/10x6_snake/wind_simulations/05ms/wind_at_cell_centers_1.csv_U Does N
ot Exists!
[ WARN] [1564687479.027731377]: [filament] No more wind data available. Using last Wind snapshopt as SteadyState.
..
[filament_simulator-2] process has finished cleanly
log file: /home/karuka/.ros/log/fa286c44-b491-11e9-9b46-3413e858c5ff/filament_simulator-2*.log

`roslaunch GADEN_player.launch`
..
[ INFO] [1564690259.227063810]: [fake anemometer]: wind noise: 0.100000
[ INFO] [1564690259.234434708]: [Player] Initializing 1 instances
[ERROR] [1564690259.298156718]: "map" passed to lookupTransform argument target_frame does not exist.
[ERROR] [1564690259.299691619]: "map" passed to lookupTransform argument target_frame does not exist.
[ERROR] [1564690259.300692760]: "map" passed to lookupTransform argument target_frame does not exist.
[ERROR] [1564690259.304641330]: "map" passed to lookupTransform argument target_frame does not exist.
[ERROR] [1564690259.339587204]: "map" passed to lookupTransform argument target_frame does not exist.

Please help me!

aravindsrj, 2020-04-17:
I too get the same issue:
[ERROR] [1586985775.413528625, 31.300000000]: "map" passed to lookupTransform argument target_frame does not exist.

PepeOjeda, 2021-02-02:
Hello, sorry for the late response. The "map" frame transform is published by the map_server node, if you are using the sample .launch files provided with gaden. If you have modified them, check the file you are using includes the lines:

<node name="map_server" pkg="map_server" type="map_server" args="$(find test_env)/$(arg scenario)/occupancy.yaml">
      <param name="frame_id" value="map"/>
</node>

, where the value of "args" should be a valid path to the .yaml file generated by the preprocessing node.

That being said, it is normal for this error to occur right at the time of launching, since all nodes are started at the same time, and the nodes that try to use the transform might start working before it has been first published. If this is the case, the error should only appear at the moment of launching, and then it dissappears and does not affect the functioning of the program in any way.

On the other hand, if the error persists, there is almost definitely something wrong with the map_server node. Could be that it's not properly installed, that it's not included in the .launch file, or that the .yaml and .pgm files that define the map do not exist.

closed 2021-02-09 by PepeOjeda