MAPIRlab/gaden #17: Problem to create a new scenario

ronnierfr, 2020-04-26:
Hello, 
First of all, I would like to congratulations on the developed GADEN package. He's sensational. I am trying to create a scenario similar to what you have developed known as "10x6_empty_room", but with reduced measures, as I need a 50x40 cm "room". I used your cad as a basis to change the scale and get an inner ".stl" file next to what I need. So I imported this file into SimScale, I managed to perform the "Mesh", but when I try to run the CFD simulation, I have problems related to the "Courant number (CFL)", because it exceeds the limit of 1. I don't know if the problem is with the settings I made for the simulation or some other. Could you help me in any way? I used the parameters suggested by you as: Boundary Conditions: Velocity Inlet (assigns random values ​​for these speeds) and Pressure Outlet and the other recommended settings. I need some help.
Thank you. Ronnier Frates Rohrich.

MSathler, 2022-10-20:
Hi @ronnierfr,

I have created a tutorial with more details, if you are interested, I've left a link below.
https://github.com/MSathler/tutorial_gaden_environment/blob/main/Gaden_Environment_Tutorial.pdf

Regards.

ronnierfr, 2022-10-20:
Hi Muricio,
Thank you very much for your attention and for sending the tutorial. This
document will be of great help in my research.

Best regards.

TaiE64, 2024-04-08:
> Hi @ronnierfr,
> 
> I have created a tutorial with more details, if you are interested, I've left a link below. https://github.com/MSathler/tutorial_gaden_environment/blob/main/Gaden_Environment_Tutorial.pdf
> 
> Regards.

Thank you so much Sir!