MAPIRlab/gaden #19: fatal error: olfaction_msgs/gas_sensor_array.h: No such file or directory  #include <olfaction_msgs/gas_sensor_array.h>

GMDFr, 2020-10-23:
Hi,
First, it's a really great job, rare are the gas simulation on ROS.

**The background :** 
I work with Gaden to implement a robot to simulate a chemical servoing. We would like to test few algorithms to locate the gas source. I installed Stage from source in my /home space.
I success to compile in a workspace dedicated :
Finished  <<< gaden_preprocessing                     [ 0.1 seconds ]          
Finished  <<< gaden_filament_simulator                [ 0.1 seconds ]          
Finished  <<< gaden_environment                       [ 0.1 seconds ]          
Finished  <<< test_env                                [ 0.1 seconds ]          
Finished  <<< olfaction_msgs                          [ 0.1 seconds ]          
Finished  <<< gaden_player                            [ 0.2 seconds ]
[build] Summary: All 6 packages succeeded!

I also be able to preprocess, GADEN.launch, GADEN_player.launch & main_simbot.launch.
But the robot doesn't move, probably due to the packages which don't compile.

**The Problem :**
When I want to compile the following package : 
gsl_actionserver
simulated_gas_sensor
simulated_gas_sensor_array

I got the following error :
fatal error: olfaction_msgs/gas_sensor_array.h: No such file or directory
 #include <olfaction_msgs/gas_sensor_array.h>

**I tried :**
- copy paste and rename fake_sensor_array.h to the include folder.
- recompile it

Did you encounter the same problem ?
How I can resolve the lacking header ?

Thank's you so much

yarraaaaa, 2021-01-31:
i have the similar problem.did u solve it? thanks

PepeOjeda, 2021-01-31:
Hello. First, please make sure you have downloaded the [olfaction messages package](https://github.com/MAPIRlab/olfaction_msgs) into your catkin workspace. If you already have, the problem may be caused by the order of compilation, since the actual header files for the message definitions (.h) are not in the package source code, and are instead auto-generated at compilation. Therefore, it is possible the headers do not yet exist at the point the other packages that need them are compiled.

The easiest solution to this is to simply run catkin_make twice in a row. The first time there might be packages that fail to compile, but the message headers will be generated, making the second compile work. 

If for some reason this still failed, try placing a CATKIN_IGNORE file inside of the packages that complain about dependencies so that you can do an error-free compilation of the message files, and then remove the CATKIN_IGNORE files for the second compilation.

GMDFr, 2021-02-01:
Hello,
We succeed to solve it and we worked on a structure more "ROS" like with the collaboration of Pepe. 
This package will be send to the GADEN Team first, to validate and will be publish open source with their permission :) 
Also, as Pepe said, the library your are looking for are on the mentionned github or on your devel folder.

 On my side, this issue is solved.

Best regards

yarraaaaa, 2021-02-01:
thanks a lot,the problem was solved by running catkin_make twice .thx for your help~

yarraaaaa, 2021-02-01:
hi,first i would say thanks a lot,the problem was solved by running catkin_make twice .thx for your help~

closed 2021-02-02 by PepeOjeda

yarraaaaa, 2021-02-03:
hello,sorry for bothering u again,i tried gaden in rviz,it's very convenient ,however,i want to use the test environment in gazebo, i'm not sure whether gaden could be used in gazebo,if it could ,how? thank u a lot .

PepeOjeda, 2021-02-04:
I'm afraid I am not familiar with Gazebo, so I can't help you with the details, but it should be possible. Once the filament simulator has already been run and the log files generated, you would have to create a new launch file that includes the gaden_player node, as well as gazebo, and then use topics to communicate with gaden_player from the code of your application.

forgetto, 2023-03-20:
Hi,
 it's a very good job, rare are the gas simulation on ROS.
I also encountered the above problems, trying the above two methods failed to solve. I would like to ask if it is possible that I changed the olfaction messages package from ros2 to ros1. I hope someone can answer it. thank you
If the original package of ros1 is available, so much the better.

PepeOjeda, 2023-03-20:
Hello. The ros1 and ros2 versions of the olfaction_msgs package are both available [here](https://github.com/MAPIRlab/olfaction_msgs) as separate branches. See the [ros1 branch](https://github.com/MAPIRlab/olfaction_msgs/tree/ros1).

forgetto, 2023-03-20:
@PepeOjeda Thank you very much. My problem has been solved