MAPIRlab/gaden #21: [WARN] No more wind data available

doublewwww, 2022-05-20:
Dear developer：
hi！
I'd  like to run the example in the test_env. when i roslaunch test_env/10x6_central_obstacle/launch/GADEN.launch after i roslaunch test_env/10x6_central_obstacle/launch/GADEN.launch.  there is the warm i got as the picture show. 
[image]

Thank you in advance for your help 
doulew

PepeOjeda, 2022-05-20:
Hello.
The message says that the wind files cannot be found. Since the path itself looks correct, I am guessing that you just need to run the GADEN_preprocessing.launch that corresponds to that environment. 

Inside of the preprocessing launch file there is a line that says: 

<param name="wind_files"  ... />

You should modify the value  of this parameter to point at the folder of the wind simulation you are trying to use (in this case, the "wind_simulations/05ms" folder).

After you have run the preprocessing, the main GADEN node should work correctly.

doublewwww, 2022-05-22:
Thank you for replying me so quickly.I have try the method you told me,there wiil be the similar warn like i first ask you.The difference is where i mark by red arrow in the picture.
 
[image]

After i run the Gaden.launch in the pkg test_env, the diffusion of gas shows in  the Rviz  last only 3 or 4 seconds and then keep steady. I guess that is the process of rapid display of diffusion. Then,i read the file named ‘filament_simulator.cpp’ in the pkg ‘gaden_filament_simulator’. There is line about why the roswarn will appear.   

[image]

So,i suppose that the warn maybe will always appear,and it is not so important.Am i right？
Thank you again for you help!

PepeOjeda, 2022-05-23:
Yes, in this case that warning is normal. It is just telling you that, since there is a single wind snapshot, it will be used as a steady state instead of reading new ones (because there aren't any more in the project).

In any case, you should be able to remove that warning to by setting the parameters "allow_looping" to true,  and "loop from step" and "loop to step" to 0 in the GADEN.launch file.

closed 2022-05-23 by PepeOjeda