MAPIRlab/gaden #22: GADEN wind flow

kynoali, 2022-05-22:
Hi 
I am using Gaden for BS my project. I want to generate my own environment and then use the GADEN c++ filament node to generate plum in my own environment. I followed all the procedures given in the gaden tutorial. when I run preprocessing launch file it is neither giving me any error nor generating an environment.  I run the simulation for 6s to generate the wind file(.csv file) and got many files. i am unable to find the problem at which step I did wrong. Can any one help me please

PepeOjeda, 2022-05-23:
Hello. I am afraid I would need more information to help you. Could you share the files that you are using, or at least the error message that you are getting?

Also, just to clarify, GADEN does not generate the airflow, only the dispersion of the gas. You are expected to provide the wind files generated with openfoam (or simscale, or whatever you use) as an input.

kynoali, 2022-05-24:
#PepeOjeda
Thanks for response, i generated wind file using OpenFOAM and simscale.  here i am going to attach some screenshots 
terminal screen short 
[image]
Preprocessing launch file 
[image]
CAD models
[image]

.CSV file (unable to upload whole file )
[image]

PepeOjeda, 2022-05-24:
There is one mistake in the launch file. The output model parameter should be named "outlets_model_0", rather than "outlets_model_1".

Still, this should not be what the issue is about. If you are not getting any messages at all, there must be some issue while first parsing the environment models. Could you share the stl files with me so I can test this?

kynoali, 2022-05-24:
Thanks sir, i send  .stl file and .CSV file on your email [Ojedamorala@hotmail.com]

closed 2022-06-03 by PepeOjeda