MAPIRlab/gaden #23: GADEN preprocessing error - Segmentation fault

rafluq, 2022-10-16:
I tried to run GADEN preprocessing on my test environment "wwt_plant" but it gives me an error. I run in Debug mode and it gives a segmentation fault in line:
https://github.com/MAPIRlab/gaden/blob/86d09bbee02c43a365725f734408e6dcc14ff322/gaden_preprocessing/src/preprocessing.cpp#L617

This happens when I have meshed the objects with a resolution of 0.2 meters, but it is fine when the mesh has a resolution of 1 meter (test environment "wwt_plant2").

Do you know why? Thank you!

I have attached both environments folders in this WeTransfer link: https://we.tl/t-PgfYfQkJ8w

PepeOjeda, 2022-10-17:
Hello. It seems like there is some problem with the wind simulation. 

The x coordinate of your environment environment goes from -16 to 16, but there are wind values recorded for points at -37 and more. Those points are outside the environment you have defined with the CAD models, so you get a segfault when processing them.

It's difficult for me to say what might have caused this. Maybe you have several versions of the 3D model that have different scale and have used another one for the simulation?
[image]
[image]

rafluq, 2022-10-19:
That was exactly the problem. Thank you so much

closed 2022-10-19 by rafluq