MAPIRlab/gaden #24: issues about GADEN_preprocessing in GADEN

17qianshi, 2022-10-18:
I want to use the GADEN to create my project, but i met some issues about GADEN_preprocessing.launch. 

I use the "gui.py" provided by GADEN to import the wind data created by FLUENT software, and CAD models built by myself. When i roslaunch the GADEN_preprocessing.launch,there is no result but no error either as the picture show.Is there something so important i ignored? Was it related to the cad models?

[image]

There is the picture about my .csv file.
[image]

 thank you

PepeOjeda, 2022-10-18:
The gui.py utility has not received any maintenance in a while, but I have just tested it on some of my files and seems to be working ok.

If you share the generated launch file I might be able to find what's wrong.

17qianshi, 2022-10-18:
there is demo work, could you help me to test it?

there are four cad models in the "cad" folder, which mean enviroment model, obstacle models, inlet model and outlet model individually.  

[new_env.tar.gz](https://github.com/MAPIRlab/gaden/files/9809793/new_env.tar.gz)

PepeOjeda, 2022-10-19:
Well, it seems like the problem is with the 3D models. There are two problems, actually.

The main one is that they are binary STL files, but have an ASCII header. Gaden can read both types of STL files, but it uses the header to determine which type of file it is. Since it sees an ASCII header, it tries to parse it as an ASCII file, and ends up in an infinite loop. I will see if I can change the parsing code to avoid this sort of problem, but in the short term the solution is to generate the files with a tool that does not mix up the formats like that.

Upon inspecting the files a bit more, I ran into another problem. I opened them with blender, and found that the scale is completely bonkers (the objects are tens of thousands of meters wide!).

I suspect whatever tool you are using to generate or export the STL files is to blame for both issues.

17qianshi, 2022-10-22:
Thank you for you helping me.

I have redrawn the models by "onshape" on web, i think i solved the problem.

Now, there is a problem about ".dae" files, so I want to ask you how you obtained the ".dae" model.

 
[image]

Thank you very much.

PepeOjeda, 2022-10-24:
If you are using onshape, you can export the .dae models from there as well. The option might be labelled "COLLADA", rather than dae, but it is the same format.

closed 2022-10-24 by PepeOjeda