MAPIRlab/gaden #27: gaden_player no wind readings (according to the anemometer) with own environment

Herwich012, 2023-06-07:
First of all, thank you for the great piece of software and documentation. It has been nice to use!

The `10x6_central_obstacle` test environment runs without issues. When I try to use my own environment resembling a small warehouse however, it seems as if the wind data is not coming through properly according to the fake anemometer that is used when running the `gaden_player` pkg, some details:
- The environment is a small warehouse (10x15x9m) occupied with filled/empty racks, it has one inlet and one outlet. The inlet (depicted top left) introduces a flow of ~1.5m/s, see picture: (taken after `roslaunch GADEN.launch`)
[image]
- I've checked the formatting of the wind data .csv, it should agree with the example.
- I've checked the `"wind_data"` param for both the preprocessing and filament simulator launch files 
- `gaden_preprocessing` seems to work as expected, it produces the '.cvs_U' etc., the occupancy.pgm/yaml and the OccupancyGrid3D.csv also seems to be ok  (although there are some little artefacts in the top left of each frame, I think they should not be a problem, or should they?)
- The `GADEN.launch` file seems to work as expected, although it might already indicate the lack of a windfield. However, because the CFD is only run until 5s simulated time it is hard to tell. In this case I've set the filament simulator to loop over the single wind data snapshot available.
- I've checked the wind data in I used in paraview, verifying that there should be significant wind velocity present at the location (2.0 0.6 1.0) of the anemometer:
[image]
- When running the `gaden_player` package the arrow depicting the anemometer is flat due to the lack of wind field data, this is confirmed by the the `ROS_INFO` printout I added to `fake_anemometer.cpp`:
`[ INFO] [1686152927.480873745]: [fake anemometer]: wind readings u v: 0.000000, 0.000000`
- Although for the 10x6_central_obstacle environment it produces:
`[ INFO] [1686154211.311332298]: [fake anemometer]: wind readings u v: 0.842310, 0.025746`
as expected

I've included a .zip of my environment, I am eager to get to the bottom of this!
[warehouse.zip](https://github.com/MAPIRlab/gaden/files/11679755/warehouse.zip)
Many thanks in advance.

PepeOjeda, 2023-06-08:
After having a look at the data it seems like the wind data was generated at a lower resolution than you are using in gaden (0.2m rather than 0.1m? I cant really say for sure), so after the preprocessing step there are "gaps" in the airflow field. 

To be fair, this is something that we could try to do something about during the preprocessing (and I'll look into it) but for now try to make sure the resolution used in gaden matches up with the one used in openfoam.

PepeOjeda, 2023-06-09:
Whoops, that crash was because of a very silly change that we (I) made at some point in the past. Fixed now, so I'll close the issue.

closed 2023-06-09 by PepeOjeda