MAPIRlab/gaden #28: Preprocessing: too many env cells due to floating point error?

Herwich012, 2023-08-25:
Hello,
I came across a weird inconsistency where the environment is divided into too many cells.
For example, when running preprocessing this is the output (with some extra printout):

```
$ roslaunch GADEN_preprocessing.launch 
... logging to #########
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server #########

SUMMARY
========

PARAMETERS
 * /preprocessing/cell_size: 0.2
 * /preprocessing/empty_point_x: 5.0
 * /preprocessing/empty_point_y: 0.6
 * /preprocessing/empty_point_z: 2.0
 * /preprocessing/model_0: #######
 * /preprocessing/number_of_models: 1
 * /preprocessing/number_of_outlet_models: 2
 * /preprocessing/outlets_model_0: #######
 * /preprocessing/outlets_model_1: #######
 * /preprocessing/output_path: #######
 * /preprocessing/wind_files: #######
 * /preprocessing/worldFile: #######
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /
    preprocessing (gaden_preprocessing/preprocessing)

auto-starting new master
process[master]: started with pid [13487]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 2f1fb62e-4321-11ee-a6d2-3ffe59c554a6
process[rosout-1]: started with pid [13513]
started core service [/rosout]
process[preprocessing-2]: started with pid [13517]
Dimensions are:
x: (-0.2, 10.2)
y: (-0.2, 16.2)
z: (0, 8)
Processing the mesh...
0%
12%
18%
43%
54%
93%
Took 0.0116553 seconds 
Processing the mesh...
0%
41%
33%
83%
66%
Processing the mesh...
0%
75%
91%
Filling...
env_size: 83 52 40
U.size():172640
V.size():172640
W.size():172640
[ INFO] [1692952030.504346063]: Preprocessing done
================================================================================REQUIRED process [preprocessing-2] has died!
process has finished cleanly
log file: ########
Initiating shutdown!
================================================================================
[preprocessing-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```
For the y-direction, the environment size is 16.4 and the cell_size is 0.2, so 82 cells are expected in the y-direction. However, due to floating point errors in this line it becomes 83 
https://github.com/MAPIRlab/gaden/blob/ccbe760f65cf8b31a667945df777744dc10c5210/gaden_preprocessing/src/preprocessing.cpp#L734
`ceil` rounds up to 83. I am not sure if this is intended behaviour. To verify this I ran a small script:

```
#include <iostream>
#include <cmath>

int main() {
    float cell_size = 0.2;
    float roundFactor = 100.0/cell_size;
    float env_min_y = -0.2;
    float env_max_y = 16.2;
    
    float y_cells_orig = ceil((env_max_y-env_min_y)*(roundFactor)/(cell_size*(roundFactor)));
    float y_cells_preceil_float = ((env_max_y-env_min_y)*(roundFactor)/(cell_size*(roundFactor)));
    
    std::cout << "y_cells_orig: " << y_cells_orig << " (should be 82!) \n";
    std::cout << "y_cells_preceil_float: \n";
    printf("%.20f", y_cells_preceil_float); std::cout << "\n";
    return 0;
}
```

output:
```
y_cells_orig: 83 (should be 82!) 
y_cells_preceil_float: 
82.00000762939453125000
```

Looking forward to hearing from you, many thanks in advance!

PepeOjeda, 2023-08-26:
Hello. 
Yes, that sort of one-over result is a known issue, but it's kind of inevitable with limited precision arithmetic. Unless this is causing some sort of problem for the simulation, I would recommend just ignoring it.

Herwich012, 2023-08-28:
Hi,
Thanks for the response. My own code relied on the fact that it would divide properly, but it will now take into account the amount of environment cells specified by GADEN, solving the issue.