MAPIRlab/gaden #29: GADEN_file###

forgetto, 2023-08-31:
Hello, I use GADEN_file, encountered some problems like asking for your help. I can use the home30 environment completely, but when I use other environments, I get an error. The picture in question is shown below.
[ WARN] [1693490820.069516246]: [filament] File /home/miracle/wzp_ws/src/GADEN_file/House11/wind_simulations/1-2,3_fast/1-2,3_fast_0.csv_U Does Not Exists!
[ WARN] [1693490820.073787891]: [filament] No more wind data available. Using last Wind snapshopt as SteadyState.
[ERROR] [1693490827.932828416]: Could not load resource [package://my_env/House11/cad_models/House11.dae]: Unable to open file "package://my_env/House11/cad_models/House11.dae".
[image]
@PepeOjeda

PepeOjeda, 2023-09-07:
Hello.
I suspect the main problem here is that ROS is failing to find the dataset package. Have you run `catkin_make` and `source devel/setup.bash` ?

If the problem persists, you might want to download the most recent version of the dataset, which has some minor modifications to the launch files, and adds support for ROS2.

forgetto, 2023-09-08:
Hello, thank you for your reply. 
First I did the catkin_make and source devel/setup.bash operations. I experimented with House30 and it worked fine. But problems arise with House1 and other experiments. I downloaded your dataset because I was running in the ROS1 environment and your update just added the ROS2 files. I can verify in House30 that there is no problem with my operation, I have a suspicion that one of the cade_mode files is corrupted during upload and download.
 Looking forward to receiving your  reply.  @PepeOjeda 
[image]
[image]
[image]
[image]

PepeOjeda, 2023-09-08:
I cannot reproduce that error with the current version of the dataset. Even if you are only using ROS1, I recommend getting the most recent version.

velonSigs, 2023-10-17:
> Hello, thank you for your reply. First I did the catkin_make and source devel/setup.bash operations. I experimented with House30 and it worked fine. But problems arise with House1 and other [...]

楼主可以加个微信交流一下不？我最近也在进行机器人主动嗅觉的研究，我的vx是15007450825

XSY159, 2024-08-22:
> Hello, I use GADEN_file, encountered some problems like asking for your help. I can use the home30 environment completely, but when I use other environments, I get an error. The picture in question is shown below. [ WARN] [1693490820.069516246]: [filament] File /home/miracle/wzp_ws/src/GADEN_file/House11/wind_simulations/1-2,3_fast/1-2,3_fast_0.csv_U Does Not Exists! [ WARN] [1693490820.073787891]: [filament] No more wind data available. Using last Wind snapshopt as SteadyState. [ERROR] [1693490827.932828416]: Could not load resource [package://my_env/House11/cad_models/House11.dae]: Unable to open file "package://my_env/House11/cad_models/House11.dae". ![图片1](https://private-user-images.githubusercontent.com/63174022/264672402-4bef778a-0b44-4c58-b292-974d0c3d8209.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQzMTUzODEsIm5iZiI6MTcyNDMxNTA4MSwicGF0aCI6Ii82MzE3NDAyMi8yNjQ2NzI0MDItNGJlZjc3OGEtMGI0NC00YzU4LWIyOTItOTc0ZDBjM2Q4MjA5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA4MjIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwODIyVDA4MjQ0MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTM4ODQ4ZDQyNGNjMzk4Y2Y1ZjVlNWE4NThmM2NjYzQyZDVlMDBhMzNmNjIxOGI3NDU0MTJiZDhiODNkZGJkNDQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.c289P50zh_uxrxZZNqFrYh_vsvHuQ6cinducMYtUX78) @PepeOjeda

您好，楼主，我也想和您加个微信交流一下，我的微信是：Xluv_4Y，谢谢！

XSY159, 2024-08-30:
> 楼主可以加个微信交流一下不？我最近也在进行机器人主动嗅觉的研究，我的vx是15007450825
想问一下这个警告是什么意思