MAPIRlab/gaden #31: about test_env

velonSigs, 2023-10-17:
Hello! 
I am a student interested in the field of robot active olfaction. 
I have carefully studied your papers related to Gaden and downloaded your open source project on GitHub. 
Is there any documentation on how to use the test_env folder? The systerm i use is ros melodic ubuntu18
By the way,the link given in the Gaden tutorial:http://mapir.isa.uma.es/work/gaden is no longer valid. 
Thank you very much!

velonSigs, 2023-10-17:
[image]

PepeOjeda, 2023-10-19:
To use the test environments you just need to launch the included launchfiles with  `roslaunch [path to file]`. 
The order in which they need to be ran is GADEN_preprocessing -> GADEN -> GADEN_player

About the link in the tutorial, yeah, we know. Remaking the tutorial is in the TO-DO list, we just haven't gotten around to it.

velonSigs, 2023-10-19:
thanks a lot
looking forward to your tutorial on this project

closed 2023-10-19 by PepeOjeda