MAPIRlab/gaden #37: get_parameter_or error

lvzhoujing, 2023-12-12:
error: no matching function for call to ‘Environment::get_parameter_or<std::string>(const char*, const char [1])’
  227 |         CAD_models[i] = get_parameter_or<std::string>(paramName.c_str(), "");

PepeOjeda, 2023-12-12:
This is a surprising error, because `get_parameter_or` is a `rclcpp::Node` member function, not something we have defined. Which version of ROS are you using?

lvzhoujing, 2023-12-13:
> This is a surprising error, because `get_parameter_or` is a `rclcpp::Node` member function, not something we have defined. Which version of ROS are you using?

Thanks for your reply! I'm using ROS2 foxy, and I'm trying changing to ROS2 humble right now.

PepeOjeda, 2023-12-13:
Humble is the version that we are using for developement (and is an LTS version), so that's probably a good idea. 
I'm looking at the foxy headers, and it seems like [that specific version](https://github.com/ros2/rclcpp/blob/humble/rclcpp/include/rclcpp/node_impl.hpp#L318) of `get_parameter_or` (the one that returns the parameter value, rather than a boolean) [did not exist](https://github.com/ros2/rclcpp/blob/foxy/rclcpp/include/rclcpp/node_impl.hpp#L242) back then, so changing to humble should fix the problem.

I'll add something to the readme about Humble being the officially supported version to avoid this sort of problem in the future.

closed 2023-12-13 by PepeOjeda