MAPIRlab/gaden #40: about coppelia_remote_api

8ijuu, 2024-04-01:
Thanks for sharing the code. I encountered some issues during compilation:
Make Error at CMakeLists.txt:39 (find_package):
-- stderr: gaden_preprocessing
CMake Error at CMakeLists.txt:39 (find_package):
  By not providing "Findcoppelia_remote_api.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "coppelia_remote_api", but CMake did not find one.

  Could not find a package configuration file provided by
  "coppelia_remote_api" with any of the following names:

    coppelia_remote_apiConfig.cmake
    coppelia_remote_api-config.cmake

  Add the installation prefix of "coppelia_remote_api" to CMAKE_PREFIX_PATH
  or set "coppelia_remote_api_DIR" to a directory containing one of the above
  files.  If "coppelia_remote_api" provides a separate development package or
  SDK, be sure it has been installed.
I have installed coppeliasim4.6.0-rev18 and the ZMQ remote API plugin, but I still can't find the coppelia_remote_apiConfig.cmake or coppelia_remote_api-config.cmake files in the root directory of coppeliasim.
Cheers

PepeOjeda, 2024-04-02:
Hello. If you want to use the coppelia scene generation, you can find the coppelia_remote_api ament package as part of our [utlis repo](https://github.com/MAPIRlab/utils/tree/ros2/Coppelia/coppelia_remote_api).

closed 2024-04-02 by PepeOjeda

PepeOjeda, 2024-04-02:
The latest version we have tested is 4.5.1, but I would assume minor version changes to be OK. 

You do not need to build the zmq api separately, it gets built by colcon through the coppelia_remote_api package.