MAPIRlab/gaden #44: [simulated_anemometer-6] [ERROR] [1712462285.338933448] [fake_anemometer]: "map" passed to lookupTransform argument target_frame does not exist.

8ijuu, 2024-04-07:
I encountered the following issue when running ros2 launch test_env main_simbot_launch.py:
ros2 launch test_env main_simbot_launch.py 
[INFO] [launch]: All log files can be found below /home/gouqin/.ros/log/2024-04-07-11-58-04-904422-gouqin-virtual-machine-10573
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [10574]
[INFO] [xterm-2]: process started with pid [10576]
[INFO] [environment-3]: process started with pid [10578]
[INFO] [player-4]: process started with pid [10580]
[INFO] [coppelia_simulator-5]: process started with pid [10582]
[INFO] [simulated_anemometer-6]: process started with pid [10584]
[INFO] [static_transform_publisher-7]: process started with pid [10621]
[INFO] [simulated_gas_sensor-8]: process started with pid [10625]
[INFO] [static_transform_publisher-9]: process started with pid [10640]
[INFO] [map_server-10]: process started with pid [10647]
[INFO] [bt_navigator-11]: process started with pid [10662]
[INFO] [planner_server-12]: process started with pid [10679]
[INFO] [controller_server-13]: process started with pid [10681]
[INFO] [behavior_server-14]: process started with pid [10685]
[INFO] [lifecycle_manager-15]: process started with pid [10687]
[robot_state_publisher-1] [INFO] [1712462285.284175971] [PioneerP3DX.robot_state_publisher]: got segment PioneerP3DX_base_link
[robot_state_publisher-1] [INFO] [1712462285.284288396] [PioneerP3DX.robot_state_publisher]: got segment PioneerP3DX_body
[robot_state_publisher-1] [INFO] [1712462285.284304591] [PioneerP3DX.robot_state_publisher]: got segment PioneerP3DX_green_button
[robot_state_publisher-1] [INFO] [1712462285.284312715] [PioneerP3DX.robot_state_publisher]: got segment PioneerP3DX_head
[robot_state_publisher-1] [INFO] [1712462285.284320115] [PioneerP3DX.robot_state_publisher]: got segment PioneerP3DX_hokuyo_front
[robot_state_publisher-1] [INFO] [1712462285.284327334] [PioneerP3DX.robot_state_publisher]: got segment PioneerP3DX_left_wheel
[robot_state_publisher-1] [INFO] [1712462285.284334148] [PioneerP3DX.robot_state_publisher]: got segment PioneerP3DX_neck
[robot_state_publisher-1] [INFO] [1712462285.284340843] [PioneerP3DX.robot_state_publisher]: got segment PioneerP3DX_red_button
[robot_state_publisher-1] [INFO] [1712462285.284347665] [PioneerP3DX.robot_state_publisher]: got segment PioneerP3DX_right_wheel
[robot_state_publisher-1] [INFO] [1712462285.284354347] [PioneerP3DX.robot_state_publisher]: got segment PioneerP3DX_screen
[player-4] [INFO] [1712462285.266799002] [gaden_player]: Initializing 1 instances
[coppelia_simulator-5] [INFO] [1712462285.276154144] [coppelia_simulator]: coppelia_root_dir: /home/gouqin/CoppeliaSim_Pro_V4_5_1_rev4_Ubuntu22_04/
[coppelia_simulator-5] [INFO] [1712462285.276249634] [coppelia_simulator]: coppelia_scene_path: /home/gouqin/gaden/install/test_env/share/test_env/scenarios/10x6_empty_room/coppeliaScene.ttt
[coppelia_simulator-5] [INFO] [1712462285.276264293] [coppelia_simulator]: coppelia_headless: 0
[coppelia_simulator-5] [WARN] [1712462285.276341068] [coppelia_simulator]: Running coppelia with command:
[coppelia_simulator-5] 	/home/gouqin/CoppeliaSim_Pro_V4_5_1_rev4_Ubuntu22_04/coppeliaSim.sh coppeliaSim /home/gouqin/gaden/install/test_env/share/test_env/scenarios/10x6_empty_room/coppeliaScene.ttt -s -vwarnings &
[coppelia_simulator-5] Warning: Ignoring XDG_SESSION_TYPE=wayland on Gnome. Use QT_QPA_PLATFORM=wayland to run on Wayland anyway.
[simulated_anemometer-6] [INFO] [1712462285.314613444] [fake_anemometer]: wind noise: 0.300000
[simulated_anemometer-6] [ERROR] [1712462285.338933448] [fake_anemometer]: "map" passed to lookupTransform argument target_frame does not exist. 
[static_transform_publisher-7] [WARN] [1712462285.282443997] []: Old-style arguments are deprecated; see --help for new-style arguments
[static_transform_publisher-7] [INFO] [1712462285.342206753] [anemometer_tf_pub]: Spinning until stopped - publishing transform
[static_transform_publisher-7] translation: ('0.000000', '0.000000', '0.500000')
[static_transform_publisher-7] rotation: ('1.000000', '0.000000', '0.000000', '0.000000')
[static_transform_publisher-7] from 'PioneerP3DX_base_link' to 'PioneerP3DX_anemometer_frame'
[static_transform_publisher-9] [WARN] [1712462285.310978995] []: Old-style arguments are deprecated; see --help for new-style arguments
[static_transform_publisher-9] [INFO] [1712462285.380241001] [pid_tf_pub]: Spinning until stopped - publishing transform
[static_transform_publisher-9] translation: ('0.000000', '0.000000', '0.500000')
[static_transform_publisher-9] rotation: ('1.000000', '0.000000', '0.000000', '0.000000')
[static_transform_publisher-9] from 'PioneerP3DX_base_link' to 'PioneerP3DX_pid_frame'
[simulated_gas_sensor-8] [INFO] [1712462285.413706021] [fake_pid]: The data provided in the roslaunch file is:
[simulated_gas_sensor-8] [INFO] [1712462285.413852107] [fake_pid]: Sensor model: 30
[simulated_gas_sensor-8] [INFO] [1712462285.413871680] [fake_pid]: Fixed frame: map
[simulated_gas_sensor-8] [INFO] [1712462285.413879838] [fake_pid]: Sensor frame: PioneerP3DX_pid_frame
[simulated_gas_sensor-8] [INFO] [1712462285.413887579] [fake_pid]: Running at 10.000000Hz
[simulated_gas_sensor-8] [ERROR] [1712462285.449524307] [fake_pid]: "map" passed to lookupTransform argument target_frame does not exist. 
[map_server-10] [INFO] [1712462285.544061872] [PioneerP3DX.map_server]: 
[map_server-10] 	map_server lifecycle node launched. 
[map_server-10] 	Waiting on external lifecycle transitions to activate
[map_server-10] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[map_server-10] [INFO] [1712462285.544402348] [PioneerP3DX.map_server]: Creating
[planner_server-12] [INFO] [1712462285.631461923] [PioneerP3DX.planner_server]: 
[planner_server-12] 	planner_server lifecycle node launched. 
[planner_server-12] 	Waiting on external lifecycle transitions to activate
[planner_server-12] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[planner_server-12] [INFO] [1712462285.641278970] [PioneerP3DX.planner_server]: Creating
[lifecycle_manager-15] [INFO] [1712462285.648135281] [PioneerP3DX.lifecycle_manager_navigation]: Creating
[bt_navigator-11] [INFO] [1712462285.689723541] [PioneerP3DX.bt_navigator]: 
[bt_navigator-11] 	bt_navigator lifecycle node launched. 
[bt_navigator-11] 	Waiting on external lifecycle transitions to activate
[bt_navigator-11] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[bt_navigator-11] [INFO] [1712462285.690051251] [PioneerP3DX.bt_navigator]: Creating
[... 217 lines cut ...]
[planner_server-12] terminate called without an active exception
[bt_navigator-11] [INFO] [1712462300.684585367] [PioneerP3DX.bt_navigator]: Completed Cleaning up
[bt_navigator-11] [INFO] [1712462300.684661489] [PioneerP3DX.bt_navigator]: Destroying bond (bt_navigator) to lifecycle manager.
[ERROR] [simulated_anemometer-6]: process has died [pid 10584, exit code -6, cmd '/home/gouqin/gaden/install/simulated_anemometer/lib/simulated_anemometer/simulated_anemometer --ros-args -r __node:=fake_anemometer --params-file /tmp/launch_params_hz2sjysp --params-file /tmp/launch_params_9kmcttmg --params-file /tmp/launch_params_7bxlrjbp --params-file /tmp/launch_params_uz_i_3qv --params-file /tmp/launch_params_kldsq069'].
[INFO] [static_transform_publisher-9]: process has finished cleanly [pid 10640]
[ERROR] [simulated_gas_sensor-8]: process has died [pid 10625, exit code -6, cmd '/home/gouqin/gaden/install/simulated_gas_sensor/lib/simulated_gas_sensor/simulated_gas_sensor --ros-args -r __node:=fake_pid --params-file /tmp/launch_params_emsja3i2 --params-file /tmp/launch_params_46s_z56y --params-file /tmp/launch_params_aiyn8n0t --params-file /tmp/launch_params_x_suuzhq --params-file /tmp/launch_params_uo8lx7is'].
[bt_navigator-11] [INFO] [1712462300.717194803] [PioneerP3DX.bt_navigator]: Destroying
[behavior_server-14] [INFO] [1712462300.728257816] [PioneerP3DX.behavior_server]: Destroying
[ERROR] [planner_server-12]: process has died [pid 10679, exit code -6, cmd '/opt/ros/humble/lib/nav2_planner/planner_server --ros-args -r __node:=planner_server -r __ns:=/PioneerP3DX --params-file /tmp/launch_params_fypyr3r2'].
[INFO] [static_transform_publisher-7]: process has finished cleanly [pid 10621]
[coppelia_simulator-5] double free or corruption (out)
[... cut ...]
[lifecycle_manager-15] [ERROR] [1712462302.528133963] [PioneerP3DX.lifecycle_manager_navigation]: Failed to change state for node: planner_server. Exception: planner_server/get_state service client: async_send_request failed.
[lifecycle_manager-15] [ERROR] [1712462302.528207281] [PioneerP3DX.lifecycle_manager_navigation]: Failed to bring up all requested nodes. Aborting bringup.
[lifecycle_manager-15] [INFO] [1712462302.560227211] [PioneerP3DX.lifecycle_manager_navigation]: Destroying lifecycle_manager_navigation
[INFO] [lifecycle_manager-15]: process has finished cleanly [pid 10687]
I'm new to ROS2 and haven't been able to solve it on my own. Can you help me, please?

8ijuu, 2024-04-07:
[image]
[image]
[image]

PepeOjeda, 2024-04-08:
You can try to debug the TF situation by viewing the TF tree with `ros2 run tf2_tools view_frames` . It should show you if there is a connection missing between any frames.

That said, the problem is most likely to do with Coppelia, as it is reponsible for the `map->odom->base_link` TFs. If you are having a lot of trouble with it, I would suggest maybe trying another robot simulator (for example, [BasicSim](https://github.com/PepeOjeda/BasicSim)) to handle the TFs and robot movement. Gaden does not have automatic scene generation for BasicSim, but it should be pretty easy to do manually by looking at the examples in the repo.

mrayh001, 2024-05-30:
Hi ,

I am also facing same issue. I am using the same coppeliasim: CoppeliaSim_Edu_V4_5_1_rev4_Ubuntu22_04
[image]

is there any update to this?or do you recommend another Coppeliasim version?

Or is there any other way to debug this?

PepeOjeda, 2024-05-31:
As first debugging step, I would recommend making sure you are not getting any errors on coppelia itself, and that the correct scene (the one generated by gaden) is playing, as the TFs are published from a script in that scene. To do that, set the "headless" parameter to false in the launch file.

In any case, in the next update to gaden I will probably add built-in support for BasicSim and start phasing out coppelia, since it seems to be causing quite a bit of trouble and I don't particularly like working with it.

XSY159, 2024-06-11:
I also encountered the same problem, have you solved it?