MAPIRlab/gaden #48: There seems to be no wind here when simulating a large-scale environment

lvzhoujing, 2024-06-19:
10x7x3:
[image]
20x20x3:
[image]
150x150x3:
[image]

The process of modeling and CFD simulation is the same, and I found that no matter how much wind speed is set, it doesn't work.
And I also want to know what the red particles represent?Is it because the particles can't flow out? But I set up the air outlets.
Looking forward to your reply!

lvzhoujing, 2024-06-20:
50x50x3:
[image]
I don't know why there are spherical particles around the gas source, and they appear again at intervals.
There is a green wind arrow in the 10x7x3 room, but it has disappeared in other rooms. And when I put in an anemometer, the wind speed I get is very small (<1m/s).

lvzhoujing, 2024-06-21:
Changing the write control to Runtime makes it better.But wind values still strange.
[image]

PepeOjeda, 2024-06-21:
Hello.  I cannot really say for sure based on the images alone, but it seems like it might be a problem with the resolution at which the wind data was generated. Make sure that you are using the same resolution in gaden as you used in whatever CFD tool you used to create the wind, as otherwise there will be areas of the environment for which no wind vector exists.

Alternatively, if all you need is simple, homogeneous wind you can set the `uniformWind` parameter in the preprocessing node to `true` and create a file that contains a single vector value for the wind (for example: 1,0,0 would be all the file contents for a wind of 1 m/s in the positive x direction)

lvzhoujing, 2024-06-24:
> Hello. I cannot really say for sure based on the images alone, but it seems like it might be a problem with the resolution at which the wind data was generated. Make sure that you are using the same [...]

I set the algorithm for generating the mesh to standard, and then the magnitude of the wind speed is normal. Thank you very much for your reply.

lvzhoujing, 2024-06-24:
> I set the algorithm for generating the mesh to standard, and then the magnitude of the wind speed is normal. Thank you very much for your reply.

It only makes a 20m house better, and a 30m house still doesn't work.
The wind profile is normal, but the wind speed value of the sensor (<1ms) and the green arrow are still small.

lvzhoujing, 2024-06-25:
> It only makes a 20m house better, and a 30m house still doesn't work. The wind profile is normal, but the wind speed value of the sensor (<1ms) and the green arrow are still small.

I changed "wind_point_inv.scale.x = 2*sqrt(pow(u,2)+pow(v,2));	  //arrow lenght" to "wind_point_inv.scale.x = wind_speed;".The green arrow seems better.