MAPIRlab/gaden #49: gas_simulations/sim1/iteration_1 does not exist

Qwxmwdncjwncc, 2024-11-27:
Hello, I've encountered a problem while running the test_env example.

### First：
(one) (base) wyx@WUYX:~/test_ws$ source install/setup.bash
(one) (base) wyx@WUYX:~/test_ws$ ros2 launch test_env gaden_preproc_launch1.py
[INFO] [launch]: All log files can be found below /home/wyx/.ros/log/2024-11-27-16-48-34-798444-WUYX-29788
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [preprocessing-1]: process started with pid [29789]
[preprocessing-1] [INFO] [1732697314.984370541] [GadenPreprocessing]: Number of models: 1
[preprocessing-1] [INFO] [1732697314.985260037] [GadenPreprocessing]: Dimensions are:
[preprocessing-1]       x : (-0.2, 10.2)
[preprocessing-1]       y : (-0.2, 6.2)
[preprocessing-1]       z : (0, 2.5)
[preprocessing-1] 
[preprocessing-1] [INFO] [1732697314.985926174] [GadenPreprocessing]: Parsing environment model: /home/wyx/test_ws/install/test_env/share/test_env/scenarios/10x6_empty_room/cad_models/10x6_walls.stl
[preprocessing-1] [INFO] [1732697314.996954670] [GadenPreprocessing]: 10%
[preprocessing-1] [INFO] [1732697314.997193341] [GadenPreprocessing]: 21%
[preprocessing-1] [INFO] [1732697314.997342712] [GadenPreprocessing]: 87%
[preprocessing-1] [INFO] [1732697314.997539750] [GadenPreprocessing]: 12%
[preprocessing-1] [INFO] [1732697314.997734304] [GadenPreprocessing]: 25%
[preprocessing-1] [INFO] [1732697314.997911913] [GadenPreprocessing]: 90%
[preprocessing-1] [INFO] [1732697315.011466709] [GadenPreprocessing]: Number of outlet models: 2
[preprocessing-1] [INFO] [1732697315.012433528] [GadenPreprocessing]: Parsing outlet model: /home/wyx/test_ws/install/test_env/share/test_env/scenarios/10x6_empty_room/cad_models/10x6_door_right.stl
[preprocessing-1] [INFO] [1732697315.026026136] [GadenPreprocessing]: 41%
[preprocessing-1] [INFO] [1732697315.026199406] [GadenPreprocessing]: 66%
[preprocessing-1] [INFO] [1732697315.027412571] [GadenPreprocessing]: 83%
[preprocessing-1] [INFO] [1732697315.039757562] [GadenPreprocessing]: Parsing outlet model: /home/wyx/test_ws/install/test_env/share/test_env/scenarios/10x6_empty_room/cad_models/10x6_door_left.stl
[preprocessing-1] [INFO] [1732697315.045900885] [GadenPreprocessing]: 25%
[preprocessing-1] [INFO] [1732697315.046023906] [GadenPreprocessing]: 83%
[preprocessing-1] [INFO] [1732697317.832755401] [GadenPreprocessing]: Preprocessing done
[INFO] [preprocessing-1]: process has finished cleanly [pid 29789]
(one) (base) wyx@WUYX:~/test_ws$ ros2 launch test_env gaden_sim_launch.py
[INFO] [launch]: All log files can be found below /home/wyx/.ros/log/2024-11-27-16-48-45-612314-WUYX-29872
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rviz2-1]: process started with pid [29876]
[INFO] [environment-2]: process started with pid [29878]
[INFO] [filament_simulator-3]: process started with pid [29880]
[environment-2] [WARN] [1732697325.840418686] [Environment]: Specifying models through numbered parameters is deprecated. You should use a single list parameter instead (see test_env for examples)
[filament_simulator-3] [WARN] [1732697325.858446443] [FilamentSimulator]: File /home/wyx/test_ws/install/test_env/share/test_env/scenarios/10x6_central_obstacle/OccupancyGrid3D.csv Does Not Exists!
[filament_simulator-3] [WARN] [1732697325.858623154] [FilamentSimulator]: File /home/wyx/test_ws/install/test_env/share/test_env/scenarios/10x6_central_obstacle/wind_simulations/1ms/wind_at_cell_centers_0.csv_U Does Not Exists!
[filament_simulator-3] [WARN] [1732697325.858634756] [FilamentSimulator]: No more wind data available. Using last Wind snapshopt as SteadyState.
[rviz2-1] QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-wyx'
[rviz2-1] [INFO] [1732697327.187996313] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-1] [INFO] [1732697327.188268042] [rviz2]: OpenGl version: 4.2 (GLSL 4.2)
[rviz2-1] [INFO] [1732697327.272407642] [rviz2]: Stereo is NOT SUPPORTED

**### Then：**
(one) (base) wyx@WUYX:~/test_ws$ source install/setup.bash
(one) (base) wyx@WUYX:~/test_ws$ ros2 launch test_env gaden_player_launch.py
[INFO] [launch]: All log files can be found below /home/wyx/.ros/log/2024-11-27-16-50-21-398143-WUYX-30223
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [xterm-1]: process started with pid [30224]
[INFO] [environment-2]: process started with pid [30226]
[INFO] [player-3]: process started with pid [30228]
[environment-2] [WARN] [1732697421.627093715] [Environment]: Specifying models through numbered parameters is deprecated. You should use a single list parameter instead (see test_env for examples)
[player-3] [INFO] [1732697421.636211738] [GadenPlayer]: Initializing 1 instances
[player-3] [ERROR] [1732697425.859132759] [GadenPlayer]: File /home/wyx/test_ws/install/test_env/share/test_env/scenarios/10x6_central_obstacle/gas_simulations/sim1/iteration_1 does not exist
[player-3] 
[player-3] [ERROR] [1732697426.369859436] [GadenPlayer]: File /home/wyx/test_ws/install/test_env/share/test_env/scenarios/10x6_central_obstacle/gas_simulations/sim1/iteration_2 does not exist
[player-3] 
[player-3] [ERROR] [1732697426.878057214] [GadenPlayer]: File /home/wyx/test_ws/install/test_env/share/test_env/scenarios/10x6_central_obstacle/gas_simulations/sim1/iteration_3 does not exist
[player-3] 
[player-3] [ERROR] [1732697427.389089895] [GadenPlayer]: File /home/wyx/test_ws/install/test_env/share/test_env/scenarios/10x6_central_obstacle/gas_simulations/sim1/iteration_4 does not exist
[player-3] 

Looking forward to your reply, thank you.

PepeOjeda, 2024-11-27:
Hello. It seems you are using a different environment to run the simulator (10x6_central_obstacle) from the one you preprocessed (10x6_empty_room), so the error is due to the corresponding environment not having been preprocessed. You can select which scenario each launch file is using by modifying the corresponding parameter at the top of the file.

Qwxmwdncjwncc, 2024-11-28:
> Hello. It seems you are using a different environment to run the simulator (10x6_central_obstacle) from the one you preprocessed (10x6_empty_room), so the error is due to the corresponding [...]

Following your guidance, I have resolved this issue. Thank you very much for your assistance!

closed 2024-11-28 by PepeOjeda