MAPIRlab/gaden #51: Simulating with Multiple Sources

sethnabarro, 2025-03-20:
I've read that GADEN supports experiments with multiple gas sources, and I can see from the documentation for the gaden player how to load and display the data for multiple sources. However, it's unclear to me how to simulate with multiple sources. Is it best to assume they are independent and simulate each source separately?

Thanks!

PepeOjeda, 2025-03-20:
Yes, that is the intended workflow. You run one simulation for each source, and store the results independently. Then, for playback, you tell the player node to load all of the simulations together.

I'd be really interested to know if you run into any issues with this, as the multi-source workflow has regrettably not been tested very much lately.

sethnabarro, 2025-03-21:
Thanks @PepeOjeda! I will let you know how it goes!

So I will create multiple simulation configs (e.g. `test_env/scenarios/Exp_C/sim1.yaml`,  `test_env/scenarios/Exp_C/sim2.yaml`, etc) and set different gas parameters in each. Then run each similutation individually and combine the results in the player.

I assume in the params file e.g. `test_env/scenarios/Exp_C/params/gaden_params.yaml` I will need to list all the sources. I think this will cause the `source_x`, `source_y`, `source_z` variables to clash between the different simulations. Is there an easy way around this or do I have to list the source coordinates in both the simulation yaml and the gaden params yaml?

PepeOjeda, 2025-03-21:
I believe the only change you have to make to `gaden_params.yaml` is to give the list of simulation results to the player node (with the [num_simulators and simulation_data_i properties](https://github.com/MAPIRlab/gaden/blob/6063fa9a5809120e4047d06ea52dfdda6af1991e/test_env/scenarios/Exp_C/params/gaden_params.yaml#L79)). 

The source location is actually not needed by the player node, although it is used by the environment node for visualization. With the current format of the params files, you would need to specify each of the locations manually inside `gaden_params`, if you want to have the source position markers to be published.

PepeOjeda, 2025-03-28:
There is currently no exposed parameter to control the color of the gas markers. I will try to add it, it should be a simple enough change.

PepeOjeda, 2025-03-28:
Done! In the most recent commit (version `2.6.2`) you can control the color of the gas with the following param:

zhouxj, 2025-03-28:
> Done! In the most recent commit (version `2.6.2`) you can control the color of the gas with the following param of the `gaden_player` node:
> 
> gas_display_color_0: [1.0, 0.0, 0.0] #red

Thank you so much!