MAPIRlab/gaden #52: How to view the gas dispersal instead of filament in RViz?

zhouxj, 2025-03-28:
Great job! 
I have installed GADEN program successfully, however the gaden_player cannot generate gas dispersal. So can you tell me how to set the launch file to get the **gas dispersal** rather than filament dispersal? 

[image]

PepeOjeda, 2025-03-28:
Hello. The second visualization corresponds to the `gaden_player` node, rather than the `filament_simulator` node. If you are using the `test_env` package, you can use it through the [gaden_player_launch.py](https://github.com/MAPIRlab/gaden/blob/ros2/test_env/launch/gaden_player_launch.py) file.

zhouxj, 2025-03-28:
Thanks a lot!