MAPIRlab/gaden #54: gaden_preproc_launch.py doesn't seem to generate the OccupancyGrid3D.csv file

NotSoCoolLikeYou, 2025-04-09:
In the tutorial it is stated that gaden_preproc_launch.py is "a launch file in charge of loading the CAD models (.stl) and CFD wind data to generate the Occupancy3D grid map", but after I run it and then run gaden_sim_launch.py, it is said that "No occupancy file provided to environment node!", like shown below in the picture. By the way I have changed the launch argument "scenario" in both launch files to "10x6_empty_room", and despite the errors the rviz visualization can still be opened, but I can only see the environment, not the gas distribution.

[image]

PepeOjeda, 2025-04-09:
You got an error when launching the preprocessing step, so it didn't run at all. It seems like you have the `generateCoppeliaScene` parameter set to `true`, but ROS cannot find `coppelia_ros2_pkg`. If you want to use it, is [here](https://github.com/MAPIRlab/utils/tree/ros2/Coppelia). Otherwise, just disable the coppelia scene generation by changing the parameter in the launch file.

NotSoCoolLikeYou, 2025-04-09:
> You got an error when launching the preprocessing step, so it didn't run at all. It seems like you have the `generateCoppeliaScene` parameter set to `true`, but ROS cannot find `coppelia_ros2_pkg`. If you want to use it, it is [here](https://github.com/MAPIRlab/utils/tree/ros2/Coppelia). If you already have compiled that package, the error might simply be that you need to source the `install/setup.bash` file.
> 
> Otherwise, just disable the coppelia scene generation by changing the parameter in the launch file.

Thanks for the fast reply! Even if I set the argument generateCoppeliaScene to False in command line, it doesn't seem to solve the problem (as shown in the figure below). However, by installing coppelia_ros2_pkg, the problem is solved, even if I actually don't need to use coppeliasim.

[image]

closed 2025-04-09 by NotSoCoolLikeYou